igrr/mkspiffs #20: Extract fails when first entry depth is over 2

bhamon, 2017-07-13:
## Problem

Let's say I have a SPIFFS which returns the following with a `mkspiffs -l` call:
```
44300   /ui/fonts/material-icons.woff2
14696   /ui/fonts/roboto-300.woff2
14748   /ui/fonts/roboto-500.woff2
14600   /ui/fonts/roboto-400.woff2
1406    /ui/favicon.ico
498     /ui/index.html
28294   /ui/main.0bd010c7.js.gz
1418    /ui/manifest.0bd010c7.js
73767   /ui/vendors.0bd010c7.js.gz
3       /config.json
```

Unpacking such a SPIFFS image fails with a `Can not create directory!!!` error.

## Analysis

After creating the root folder, the program tries to create `/ui/fonts/` with a single `dirCreate` call.

## Correction

I see two solutions :
- (Better approach) Recursively creates directories by cutting the path to pieces.
- Ensure files are inserted before directories when building SPIFFS.

igrr, 2017-08-22:
Good catch, thanks! I think the second option may be unreliable if you have a structure similar to:
```
/dir_a/dir_b/file_1
/dir_a/dir_b/file_2
/file3
```
The first option should work though. Are you interested in sending a PR for this? :)

bhamon, 2017-08-24:
Yes, I'll create a PR ASAP based on the first solution.

igrr, 2017-09-04:
Thanks for the PR!

closed 2017-09-04 by igrr