igrr/mkspiffs #26: 0.3.7 produces corrupt images

universam1, 2017-09-15:
Tested with latest Arduino where 0.3.7 is the library. Files are existing but have 0 bytes.

Any idea?

igrr, 2017-09-15:
Not really... @me-no-dev can you check this?

me-no-dev, 2017-09-15:
this must be coupled with the new mkspiffs as well. Older SPIFFS partitions might not work, because there are some new security features (confirms that the size of the partition is actually correct). If you have old SPIFFS partition, I recommend format

igrr, 2017-09-15:
Huh? Why, that's a showstopper, we should be able to use old images with new firmware, otherwise folks will not be able to use OTA to update, say, from core version 2.3.0 to 2.4.0. Do you know which new features are these? Maybe we can disable them for backwards compatibility.

universam1, 2017-09-15:
Exactly, thats the showstopper I faced on a production system with OTA out of a sudden.

me-no-dev, 2017-09-15:
https://github.com/igrr/mkspiffs/blob/master/spiffs/spiffs_config.h#L202

me-no-dev, 2017-09-15:
so turn this off on ESP8266 and ESP32?

me-no-dev, 2017-09-15:
TBH I think this feature is quite good to be kept on ;) both chips (especially ESP8266) can be misconfigured, which would lead to partition being mounted, but from then on it's unknown. Here is a scenario: you flash with the wrong setting and partition will mount, but your data is or will be corrupted.

igrr, 2017-09-15:
I get that this is a very useful option, but i don't think that breaking backwards compatibility with previous releases of Arduino is okay, in a minor version change (2.3.0->2.4.0). We can change this option if the next release is a major bump (e.g. 3.0.0).

For ESP32 we may as well keep that option enabled. This would require having different mkspiffs builds for ESP32 and ESP8266, but that is not so hard to do.

MMBwim, 2017-10-06:
Hi guys, just a HW engineer, still, i am also stuck at the same point i think.
I am trying to build a simple webserver that reads the files from SPIIF.
my status: 
- mkspiff with magic enable can be mounthed, seems to be working, files get corrupted as it seems, reading strange contaent back forever.
if i do a write towards that file seems to be working, reading still corrupted.
- mkspiff without magic enabled, can't be mounthed.

Any luck? 

Thanks, Wim

me-no-dev, 2017-10-08:
@MMBwim which chip are you working with? ESP8266 or ESP32?

MMBwim, 2017-10-08:
ESP32

MMBwim, 2017-10-08:
ESp32Thing Sparkfun

MMBwim, 2017-10-08:
P.S. Spiff example works, mkspiff seems to be the problem

me-no-dev, 2017-10-08:
@MMBwim fix is in PR: https://github.com/igrr/mkspiffs/pull/31

MMBwim, 2017-10-09:
me-no-dev, 

If i understand correctly i need to adopt the changes as described in PR#31 to build MKSPiff

Thas a bit hard for me to do, i downloaded your latest release (https://github.com/igrr/mkspiffs/releases), could you make a new compiled release version including the #31 fixes? Sorry for all the questions

Thanks, WIm

me-no-dev, 2017-10-13:
@igrr can we get a new release please?

me-no-dev, 2017-10-13:
Time to close this :)

MMBwim, 2017-10-15:
Thanks! Going to test it

MMBwim, 2017-10-15:
Works, Thanks !!!!!

closed 2018-02-12 by igrr