igrr/mkspiffs #33: option -s doesn't work

nottledim, 2017-10-17:
If specify partition size with -s an image is built which cannot be read.  With -i it prints total and used as 0 and 255. The -l option causes seg fault.

It builds image OK when -s not specified but size is too small.

It's version V3.0V3.0-dev-806-gde750e99

Linux system GCC version 6.3.

(Sorry about brevity, men dug up my internet connection and I have to use my phone 😡.)

davruet, 2018-02-05:
Same problem here, (using mkspiffs git d79bd560) - if I omit size everything works, but adding -s seems to break things:

$ mkspiffs -p 256 -b 4096 -s 0x16F000 -c spiffs.img
/conf.json
$ mkspiffs/mkspiffs -d 5 -l spiffs.img
Debug output enabled
Segmentation fault: 11

OSX 10.11.3, compiled with clang++ (clang-703.0.31)

igrr, 2018-02-12:
Hi @nottledim, could you explain what you mean by "image is built which can not be read". Are you using mkspiffs to read the image, or something else?

@davruet Thanks for reporting the segfault, will fix.

nottledim, 2018-02-12:
I was building a filesystem for uploading to esp32 (esp-idf).  If I used the -s option it creates an image but that image is not useful on the esp32.  Files supposedly included in the spiffs image could not be accessed by the esp32.  I was just using code derived from the example code which reports some meta data from the filesystem.  It was that that reported the silly numbers.

One thing I recently discovered is that the default settings in esp-idf have been changed.  Specifically the "size of per-file metadata".  mkspiffs defaults to zero but esp-idf now sets I think 4 bytes (1 word).

I think it's possible that this explains the problem I was seeing.  A more recent use of latest mkspiffs with the settings matching seems to work properly.

igrr, 2018-02-12:
@nottledim ok, thanks, it makes sense.

I am working now on preparing new mkspiffs release which should be compatible with ESP-IDF. I'll update this issue when it is done.

nottledim, 2018-02-12:
Just for the record this is where I was alerted to the change: https://github.com/espressif/esp-idf/issues/1437

igrr, 2018-02-12:
Please give the latest release a try: https://github.com/igrr/mkspiffs/releases. Since you are using esp-idf, download mkspiffs-0.2.2-esp-idf-YOUR_OS archive.

closed 2018-02-12 by igrr