igrr/mkspiffs #34: Getting error (-10010) while trying to pack a simple gzipped webpage

Eloquence4, 2017-11-01:
Hello,

I am trying to pack a webpage into an image file, so that i can upload it to my esp32. This is what i try to run: mkspiffs -c ./web -b 4096 -p 256 -s 524288 web.img -d 5
As a result i get "SPIFFS_write error(-10010): unknown" for every file but the index.html. (it's the smallest file from the bunch, i don't know if the issue has anything to do with file size though)

I have tried running the command with mkspiffs 0.2.1 64bit linux, 32bit linux and 0.2.0 64 bit. No luck in any of the versions.

This is a screenshot of what i am getting: https://i.imgur.com/1FJw6hP.png

asdofe, 2018-01-12:
I had the same problem as well. Does anybody know how to solve?

igrr, 2018-02-12:
@Eloquence4 From the screenshot, it seems that the file name is too long. By default, SPIFFS is configured with the maximum file name of 32 bytes. `/static/images/apple-touch-icon.png` is 36 bytes, other files which can not be added have even longer names. The manifest file has a short name and is added successfully.

You can either shorten the names, or increase the maximum name length by overriding `SPIFFS_OBJ_NAME_LEN` parameter. See https://github.com/igrr/mkspiffs/tree/0.2.2#spiffs-configuration for instructions. If you decide to increase the maximum name length, make sure you make the same change of `SPIFFS_OBJ_NAME_LEN` on the firmware side. With ESP-IDF, you can set `SPIFFS_OBJ_NAME_LEN` in menuconfig. With Arduino, you don't have such an option, unless you use Arduino as an ESP-IDF component.

closed 2018-02-12 by igrr