igrr/mkspiffs #38: mkspiffs not working with the last espressif/esp-idf version

mmvwk01, 2018-01-23:
hi,

i use the last espressif/esp-idf  with the spiffs (spiffs @ f5e26c4).
I try to make spiffs image with the tools (i tested all versions)

I (51) boot: ## Label            Usage          Type ST Offset   Length
I (95) boot:  5 storage          Unknown data     01 82 002f0000 000f8000

./mkspiffs -c ./data -b 8192 -p 256 -s 0xFB000 ./spiffs.bin
 esptool.py --port /dev/ttyUSB0 -b 921600  write_flash 0x002f0000 ./spiffs.bin

result:
I (353) spiffs: Initializing SPIFFS
W (383) SPIFFS: mount failed, -10025. formatting...

lszlfrng, 2018-01-27:
Hi,
yesterday I had a similar problem. I remembered that it works well with the arduino-esp32. 
In the arduino-esp32 the SPIFFS_META_LENGTH is 0.
I changed the default setting in my application for the SPIFFS_META_LENGTH parameter to 0.
Please check it yourself too:
- start make menuconfig; 
- go to [Component config][SPIFFS Configuration];
- change the SPIFFS_META_LENGTH (from default 4?) to 0; 
- build your application, flash and check the result.
With this modification, I can generate spiffs partition with mkspiffs, After uploading it the application can mount and modify the content. The download and unpack is works also.
I hope I could help you.
Laszlo
P.S.: I am not a programmer so the solution may seem simple.

mmvwk01, 2018-01-30:
Hi, thanks for answer,

i set the SPIFFS_META_LENGTH to 0 in my App, but the result is the same:
W (407) SPIFFS: mount failed, -10025. formatting...

lszlfrng, 2018-02-01:
Hi,
This is not good news because it works for me. There is some difference between the two environments.
I'm working on Windows and I use the latest mkspiffs binary (https://github.com/igrr/mkspiffs/releases/tag/0.2.1, with magic length) and I have configured one single spiffs partition (spiffs, data, spiffs, 0x291000,0x16F000,).
If you send your relevant codes, I'll try and find out the root of the problem. My config screenshot has been attached.
[image]

mmvwk01, 2018-02-10:
hi
1. git clone https://github.com/espressif/esp-idf
2. cd esp-idf/examples/storage/spiffs
3. edit partitions_example.csv 
4. storage,  data, spiffs,  0x291000, 0xF0000,
5. make erase_flash
6. make flash

1. git clone https://github.com/igrr/mkspiffs
2. git submodule init
3. git submodule update
4. make
5. create dir "data"
6. create file "test.txt" in dir "data"
7. ./mkspiffs -c ./data -b 8192 -p 256 -s 0x000f0000 ./spiffs.bin
8. esptool.py --port /dev/ttyUSB0 -b 921600  write_flash 0x00291000 ./spiffs.bin

Result:
W (407) SPIFFS: mount failed, -10025. formatting...

igrr, 2018-02-12:
@mmvwk01 ESP-IDF sets block size to the size of flash erase sector (4096 bytes), whereas you are setting it to 8192.

Please try the latest release of mkspiffs from https://github.com/igrr/mkspiffs/releases/tag/0.2.2, download the version for your OS which has 'esp-idf' in its name.

For the partition table in the unmodified ESP-IDF example, you should use the following command:
```
./mkspiffs -c /path/to/source/dir -b 0x1000 -p 0x100 -s 0xF0000 spiffs_image.bin
```

Before doing this, make sure that you use default configuration of mkspiffs example in ESP-IDF. That is,
```
rm sdkconfig
make defconfig
```

closed 2018-02-12 by igrr