igrr/mkspiffs #43: libc++abi.dylib: terminating with uncaught exception of type std::length_error: vector

jeraymond, 2018-03-09:
On macOS I’m attempt to uploadfs using PlatformIO. I run `pio run -t uploadfs` after creating the data directory (containing a JSON file) and after successfully uploading my program to a Wemos D1 Mini Pro board (ESP8266).

I an exception when I run this. From the verbose output I see pio running the below mkspiffs command.

```
"mkspiffs" -c data -p 256 -b 8192 -s -1069056 .pioenvs/d1_mini_pro/spiffs.bin
libc++abi.dylib: terminating with uncaught exception of type std::length_error: vector
*** [.pioenvs/d1_mini_pro/spiffs.bin] Error -6
```

Any idea what might be wrong or how to troubleshoot this? I’m using PlatformIO version 3.5.2rc4.

igrr, 2018-03-09:
That is most likely caused by the fact that negative partition size (`-s -1069056`) is given. I think mkspiffs should check and give a better error message here, but you still need to find what calls `mkspiffs` and why the size is negative.

jeraymond, 2018-03-09:
I've found a corresponding bug on platformio/platform-espressif8266 [issue #69](https://github.com/platformio/platform-espressif8266/issues/69). It is said to be fixed in an upstream development branch (newer than current 1.6.0 release).

I was able to work around the problem by using a smaller than 16MB [custom flash size](http://docs.platformio.org/en/latest/platforms/espressif8266.html?highlight=spiffs#platform-espressif-customflash) by adding these builds flags in my `platformio.ini` file:

```
[env:d1_mini_pro]
build_flags = -Wl,-Teagle.flash.4m.ld
```

Do you want to keep this issue open for fixing the error message?

igrr, 2018-03-22:
Error message added in 011c110.

closed 2018-03-22 by igrr