igrr/mkspiffs #45: Make error windows - help please

zafrirron, 2018-03-30:
$ make dist
mkdir -p spiffs_t
cp spiffs/src/*.h spiffs_t/
cp spiffs/src/*.c spiffs_t/
rm -rf spiffs_t/.git
rm -f spiffs_t/.DS_Store
ls -1 spiffs_t > out.list0
touch spiffs_t/.DS_Store
mkdir -p spiffs_t/.git
touch spiffs_t/.git/foo
./mkspiffs -c spiffs_t -s 0x100000 -p 512 -b 0x2000 out.spiffs_t | sort | sed s/^\\/// > out.list1
./mkspiffs -u spiffs_u -s 0x100000 -p 512 -b 0x2000 out.spiffs_t | sort | sed s/^\\/// > out.list_u
./mkspiffs -l -s 0x100000 -p 512 -b 0x2000 out.spiffs_t | cut -f 2 | sort | sed s/^\\/// > out.list2
diff --strip-trailing-cr out.list0 out.list1
1,7d0
< spiffs.h
< spiffs_cache.c
< spiffs_check.c
< spiffs_gc.c
< spiffs_hydrogen.c
< spiffs_nucleus.c
< spiffs_nucleus.h
make: *** [Makefile:130: test] Error 1

igrr, 2018-03-30:
Can you please run `cat out.list0` and `cat out.list` commands and post the output?

zafrirron, 2018-03-30:
Sorry broke my environment ....
diff not found error...

igrr, 2018-03-30:
Okay, closing for now, please reopen if you can supply more info.

closed 2018-03-30 by igrr