igrr/mkspiffs #47: SPIFFS_ERR_NOT_A_FS

timredfern, 2018-04-12:
I'm trying to use mkspiffs to build a spiffs image to put some files onto ESP32, I must be doing something wrong, because mkspiffs can't read it's own image.

```
tMac:ESP tim$ git clone git@github.com:igrr/mkspiffs.git
Cloning into 'mkspiffs'...
remote: Counting objects: 307, done.
remote: Compressing objects: 100% (24/24), done.
remote: Total 307 (delta 16), reused 22 (delta 9), pack-reused 274
Receiving objects: 100% (307/307), 210.27 KiB | 893.00 KiB/s, done.
Resolving deltas: 100% (164/164), done.
tMac:ESP tim$ cd mkspiffs
tMac:mkspiffs tim$ git submodule init
Submodule 'spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'spiffs'
tMac:mkspiffs tim$ git submodule update --recursive
Cloning into '/Users/tim/ESP/mkspiffs/spiffs'...
Submodule path 'spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c'
tMac:mkspiffs tim$ make
c++ -std=gnu++11 -Os -Wall -mmacosx-version-min=10.7 -arch i386 -arch x86_64 -stdlib=libc++  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-1-g011c110\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__   -c -o main.o main.cpp
cc -std=gnu99 -Os -Wall -mmacosx-version-min=10.7 -arch i386 -arch x86_64  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-1-g011c110\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__   -c -o spiffs/src/spiffs_cache.o spiffs/src/spiffs_cache.c
cc -std=gnu99 -Os -Wall -mmacosx-version-min=10.7 -arch i386 -arch x86_64  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-1-g011c110\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__   -c -o spiffs/src/spiffs_check.o spiffs/src/spiffs_check.c
cc -std=gnu99 -Os -Wall -mmacosx-version-min=10.7 -arch i386 -arch x86_64  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-1-g011c110\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__   -c -o spiffs/src/spiffs_gc.o spiffs/src/spiffs_gc.c
cc -std=gnu99 -Os -Wall -mmacosx-version-min=10.7 -arch i386 -arch x86_64  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-1-g011c110\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__   -c -o spiffs/src/spiffs_hydrogen.o spiffs/src/spiffs_hydrogen.c
cc -std=gnu99 -Os -Wall -mmacosx-version-min=10.7 -arch i386 -arch x86_64  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-1-g011c110\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__   -c -o spiffs/src/spiffs_nucleus.o spiffs/src/spiffs_nucleus.c
c++ main.o spiffs/src/spiffs_cache.o spiffs/src/spiffs_check.o spiffs/src/spiffs_gc.o spiffs/src/spiffs_hydrogen.o spiffs/src/spiffs_nucleus.o -o mkspiffs -mmacosx-version-min=10.7 -arch i386 -arch x86_64 -stdlib=libc++ 
strip mkspiffs
tMac:mkspiffs tim$ ./mkspiffs  -c include -b 8192 -p 256 -s 1048576 ./spiffs_image.img
/spiffs_config.h
tMac:mkspiffs tim$ ls -la spiffs_image.img 
-rw-r--r--  1 tim  staff   1.0M 12 Apr 15:52 spiffs_image.img
tMac:mkspiffs tim$ ./mkspiffs  -l ./spiffs_image.img
SPIFFS mount failed with error: -10025
error: failed to mount image
tMac:mkspiffs tim$ grep -rn "\-10025" ./ --include *.h
./spiffs/src/spiffs.h:43:#define SPIFFS_ERR_NOT_A_FS             -10025
```
I'm seeing the same error on the ESP32 since I updated esp-idf the other day. I suspect that theres a regression in spiffs itself, unless I'm missing something ^^^^ ?

timredfern, 2018-04-16:
Further to this:

I had a working project for ESP32 with mkspiffs set up as a component, and make targets to flashfs etc.. I copied it from https://github.com/loboris/ESP32_spiffs_example

I'm trying to figure out why it stopped working, I think it was when I updated esp-idf, but there's something that doesn't make sense.

The working version of mkspiffs from this repo also fails the sanity check of reading it's own image.

igrr, 2018-04-17:
The test from your original post fails because you're not specifying the same parameters in the second invocation of mkspiffs. Specifically, block size is set to 8192 in the first invocation and 4096 (default value) in the second, hence SPIFFS is unable to load the image.
Calling
```
./mkspiffs  -l -b 8192 ./spiffs_image.img
```
should work.

Next, regarding using mkspiffs with ESP-IDF. You need to make sure that the compile time configuration on the `mkspiffs` side matches `sdkconfig` in your IDF application. I have tried explaining this in https://github.com/igrr/mkspiffs#spiffs-configuration.
For example, if `mkspiffs` is built with `SPIFFS_OBJ_META_LEN=0`, but your IDF app uses `CONFIG_SPIFFS_META_LENGTH=4`, then the IDF application will not be able to load the SPIFFS image generated by `mkspiffs`.
Run `mkspiffs --version` and `grep SPIFFS sdkconfig`, and compare the settings.

timredfern, 2018-04-18:
Thanks for the info on telling mkspiffs the parameters of the image file. That worked.

I'm still baffled as to how to make a mkspiffs image work with the latest versions of esp-idf.

I spent a while on the makefile and putting the latest mkspiffs and spiffs into an esp-idf project similar to https://github.com/loboris/ESP32_spiffs_example 

I set the compiler options for mkspiffs:

```
tMac:template-spiffs-debug tim$ components/mkspiffs/src/mkspiffs --version
mkspiffs ver. 0.2.3-1-g011c110
Build configuration name: generic
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: UNAVAILABLE
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 4
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0
```
and in sdkconfig:

```
#
# SPIFFS Configuration
#
CONFIG_SPIFFS_MAX_PARTITIONS=3

#
# SPIFFS Cache Configuration
#
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
CONFIG_SPIFFS_CACHE_STATS=
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
CONFIG_SPIFFS_GC_STATS=
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=y
```
and

```
#
# SPIffs Example Configuration
#
CONFIG_SPIFFS_BASE_ADDR=0x180000
CONFIG_SPIFFS_SIZE=1048576
CONFIG_SPIFFS_LOG_BLOCK_SIZE=8192
CONFIG_SPIFFS_LOG_PAGE_SIZE=256
```

And the project is set to use/write a partition at 0x180000

Now, I can read the archive with mkspiffs 

```
tMac:template-spiffs-debug tim$ components/mkspiffs/src/./mkspiffs  -l -b 8192 build/spiffs_image.img 
6224	/esp_now.rst
101	/esp_smartconfig.rst
1103	/esp_wifi.rst
223	/index.rst

```

However, the ESP32 still fails to read the partition and formats it.

igrr, 2018-04-19:
I have not tried https://github.com/loboris/ESP32_spiffs_example, but the storage/spiffs example in IDF should work with mkspiffs. Could you try that one?

timredfern, 2018-04-20:
Hi Ivan, this is the minimal test case:

```
#include "esp_vfs.h"
#include "esp_log.h"
#include "esp_spiffs.h"

static const char TAG[] = "SPIFFS test";

int app_main(void)
{
    ESP_LOGI(TAG, "starting app");

    esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = NULL,
      .max_files = 5,
      .format_if_mount_failed = false
    };

    esp_err_t ret = esp_vfs_spiffs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ESP_LOGE(TAG, "Failed to find SPIFFS partition");
        } else {
            ESP_LOGE(TAG, "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
        }
        return;
    }

    return 0;
}
```
with mkspiffs:

```
tMac:template-spiffs-debug tim$ components/mkspiffs/src/mkspiffs --version
mkspiffs ver. 0.2.3-1-g011c110
Build configuration name: generic
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: UNAVAILABLE
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 4
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0
tMac:template-spiffs-debug tim$ components/mkspiffs/src/mkspiffs -l -b 8192 build/spiffs_image.img 
6224	/esp_now.rst
101	/esp_smartconfig.rst
1103	/esp_wifi.rst
223	/index.rst
```
And in the sdkconfig:
```
#
# SPIffs Example Configuration
#
CONFIG_SPIFFS_BASE_ADDR=0x180000
CONFIG_SPIFFS_SIZE=1048576
CONFIG_SPIFFS_LOG_BLOCK_SIZE=8192
CONFIG_SPIFFS_LOG_PAGE_SIZE=256
```
and

```
#
# SPIFFS Cache Configuration
#
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
CONFIG_SPIFFS_CACHE_STATS=
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
CONFIG_SPIFFS_GC_STATS=
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=y
```

after `make flash` and

```
tMac:template-spiffs-debug tim$ make flashfs
Making spiffs image ...
python /Users/tim/ESP/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/tty.usbserial-0050131AB --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect
/esp_now.rst
/esp_smartconfig.rst
/esp_wifi.rst
/index.rst
esptool.py v2.3.1
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1048576 bytes to 5337...
Wrote 1048576 bytes (5337 compressed) at 0x00180000 in 0.1 seconds (effective 130501.0 kbit/s)...
Hash of data verified.
```

However from the console:

```
I (28) boot: ESP-IDF v3.1-dev-743-g6c44fc70-dirty 2nd stage bootloader
I (28) boot: compile time 11:46:05
I (54) boot: Enabling RNG early entropy source...
I (54) boot: SPI Speed      : 40MHz
I (54) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 4MB
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00180000 00100000
I (101) boot: End of partition table
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05a34 ( 23092) map
I (122) esp_image: segment 1: paddr=0x00015a5c vaddr=0x3ffb0000 size=0x021bc (  8636) load
I (126) esp_image: segment 2: paddr=0x00017c20 vaddr=0x40080000 size=0x00400 (  1024) load
I (132) esp_image: segment 3: paddr=0x00018028 vaddr=0x40080400 size=0x07fe8 ( 32744) load
I (154) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x16b7c ( 93052) map
I (187) esp_image: segment 5: paddr=0x00036b9c vaddr=0x400883e8 size=0x006d0 (  1744) load
I (188) esp_image: segment 6: paddr=0x00037274 vaddr=0x400c0000 size=0x00000 (     0) load
I (199) boot: Loaded app from partition at offset 0x10000
I (200) boot: Disabling RNG early entropy source...
I (206) cpu_start: Pro cpu up.
I (209) cpu_start: Starting app cpu, entry point is 0x40080e3c
I (0) cpu_start: App cpu up.
I (220) heap_init: Initializing. RAM available for dynamic allocation:
I (227) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (233) heap_init: At 3FFB29B8 len 0002D648 (181 KiB): DRAM
I (239) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (245) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (252) heap_init: At 40088AB8 len 00017548 (93 KiB): IRAM
I (258) cpu_start: Pro cpu start user code
I (276) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (277) SPIFFS test: starting app
E (277) SPIFFS: mount failed, -10025
E (277) SPIFFS test: Failed to mount filesystem
```

Sorry for the verbosity, but I think this covers everything that affects spiffs?

igrr, 2018-04-20:
Thanks for all the details. I think that in IDF, SPIFFS component always uses block size of 4096 bytes.

I could not find references to `CONFIG_SPIFFS_LOG_BLOCK_SIZE` in IDF, so I assume it comes from https://github.com/loboris/ESP32_spiffs_example/blob/master/main/Kconfig.projbuild#L16.
Note that this example also comes with its own copy of SPIFFS (it doesn't use the one in IDF), so the configuration options for that example are not applicable when you are using `storage/spiffs` example from IDF, or a test application like you have shown.

One more thing, you haven't attached your partition table. Please check that SPIFFS partition size in the partition table matches the size you pass to mkspiffs.

I was able to get your attached test code to work after changing `mkspiffs` block size argument to 4096, changing image size to 0xF0000, and using the partition table from storage/spiffs example in IDF.

timredfern, 2018-04-20:
Hi Ivan,

Yes you're correct, I'm trying to use the kconfig stuff from the loboris project to be able to use mkspiffs from the makefile. I am struggling to understand the whole system.

Changing to 4096 block size does fix the problem, with the latest esp-idf. However, the 8192 block size did used to work. I'm happy to use the 4096 block size.

Thanks for all your help.

closed 2018-04-20 by timredfern