igrr/mkspiffs #48: image comes up empty

ef1986, 2018-04-19:
Dear Mr. Grokhotkov,

I use the mkspiffs software to create images containing text files. The images are then flashed onto an ESP32. If the mkspiffs software is running on macOS (10.12.6) everything works perfectly. If the software is running on Ubuntu 17.10 however, there are no errors either(neither during image creation, nor when the ESP32 uses them), but the text files come up empty. Do you have any idea, what the problem could be? Thanks for your help in advance.

Best regards

ef1986, 2018-05-23:
Solved: Wrong build settings

closed 2018-05-23 by ef1986

snailwft, 2018-06-12:
Dear @ef1986 
I have your same problem. How did you solve it?

Best regards

ef1986, 2018-06-22:
Dear @snailwft,

Sorry for the delay in ansering! I had a lot to. If you want to use the program in combination with

make dist   BUILD_CONFIG_NAME="-esp-idf" CPPFLAGS="-DSPIFFS_OBJ_META_LEN=4
That did the trick for me.

Best regards