igrr/mkspiffs #51: Replacing 'diff --strip-trailing-cr'

b0wter, 2018-06-18:
I am converting most of my toolchains to work in docker containers and the linux distro of choice for me is alpine because of it's small size (4mb base image). However, since it uses busy box for the core components there are usually some workaround required to get things to work.

While trying to get this to work in an alpine container I found that ```diff``` doesnt support the ```--strip-trailing-cr``` parameter. Do you see any difficulties in replacing this with a ```sed```-command and run ```diff``` afterwards?

igrr, 2018-06-18:
No problem, as long as sed command works with BSD version of sed as well.

igrr, 2018-06-28:
Merged #52, thanks!

closed 2018-06-28 by igrr