igrr/mkspiffs #58: Fails to compile in OSX Mojave

kendersec, 2018-12-24:
I'm unable to compile without errors

> ld: warning: The i386 architecture is deprecated for macOS (remove from the Xcode build setting: ARCHS)
> ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd, missing required architecture i386 in file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd
> ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libc++.tbd, missing required architecture i386 in file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libc++.tbd
> Undefined symbols for architecture i386

It seems i386 has now been removed in OSX - removing it from the Makefile builds successfully.

thomasnordquist, 2018-12-30:
I can also confirm this issue. Removing the i386 architecture helped.

thomasnordquist, 2018-12-30:
The command `uname -r` yields `18.2.0` as operating system release on my machine (running Mojave).
*Just for documentation purposes*

hmxmghl, 2019-06-26:
Confirming the solution proposed in above comments.