igrr/mkspiffs #63: SPIFFS_write error(-10001): File system is full.

nospam2k, 2019-03-15:
Compiled sketch from Arduino IDE for ESP-01 (1m) with Generic ESP8266 Module and Flash Size 1M (192K SPIFFS).

 mkspiffs -c data/ firmware.bin

result:

SPIFFS_write error(-10001): File system is full

I tried changed to 256K with same error. Works from IDE using ESP8266 Sketch Data Upload plugin.

igrr, 2019-03-15:
You might need to specify a few more command line arguments, at the very least the fliesystem size. Default value is 64kB, so probably your data does not fit? Arduino IDE passes a few more arguments, such as page size and block size. You can find them in boards.txt of esp8266/Arduino repository.

nospam2k, 2019-03-15:
I've done some homework since my post and I got it. I didn't know about the default size but I did get it to work once I set the size. My problem now (I don't think it's the bin file) is I can't seem to figure out how to uploaded the spiffs.bin using OTA. I'm getting : ERROR[10]: Magic byte is wrong, not 0xE9. Thanks for the quick reply!

t5b6de, 2019-03-23:
Have same error. Sometimes i do have unknown error.
This is my output from mkspiffs:
`Debug output enabled
/assets/js/jquery.js.gz
file size: 30212
/assets/js/main.js.gz
file size: 1062
/assets/js/materialize.js.gz
file size: 71631
/assets/js/modal.js.gz
file size: 1168
/assets/main.css.gz
file size: 75
/assets/materialize.css.gz
file size: 21387
/assets/micons.css.gz
file size: 278
/assets/micons.woff2.gz
file size: 60936
SPIFFS_write error(-10001): File system is full.

data left: 264
error adding file!

Error for adding content from assets!
/config/main.json
file size: 124
/index.html.gz
file size: 782
SPIFFS_write error(-10001): File system is full.

data left: 270
error adding file!
`
whem i'm choosing bigger (512k) flash image sizze instead of 256k 
my esp8266 is not able to see this files.

at the moment i'm flashing a ftp-server to my esp, format the spiffs,  and then i upload the files to my esp. When i'm done, i am flashing my firmware which i am working on.

then it is working.

In the attachment you'll find a zip with data dir (this files are going to be on the esp)
mkspiffs.bin (file which is created with the output above)
esp_spiffs.bin (file which was read from an working esp8266 via esptools.py)

btw: Same result with die Arduino Sketch Data Upload which is using the same tool.

[spiffs.zip](https://github.com/igrr/mkspiffs/files/2999391/spiffs.zip)

I hope insanely that there is a better solution than i am working with.

kind regards.