igrr/mkspiffs #72: mkspiffs tool does not unpack image files ...

sfranzyshen, 2020-01-29:
I'm guessing not many people have tried to use this utility other than to create files ...

I have tried using the mkspiffs tool (the one packaged here in releases ... as well as the ones packages with the Arduino/Esp8266 core (v2.5.2 - v2.6.3)) to unpack the file I just created (using the arduion ide toolchain) while the spiffs image works fine after uploading it to the device ... I'm not able to see all the files (only shows the first 3 files) when I use the utility to either --list or --unpack the files ... and 'YES' i'm using the same version of the tool to both create and (try) to unpack ...

Has anyone else used this utility to 'unpack'?

igrr, 2020-01-29:
There is a basic test for the unpacking function which runs as part of `make dist`: https://github.com/igrr/mkspiffs/blob/983970e40ff381d95d68a9bddff70c4d9921021b/Makefile#L127. But of course it is entirely possible that something *is* broken.

Maybe you can try to create the image using the command line tool and then unpack the image, and see if it works? If it does, maybe there is some difference in the parameters which the IDE uses when it invokes the tool. If it doesn't, then please provide the instructions to reproduce.

Also I regret to say that I am not actively maintaining this tool anymore, as for my present use case there is a Python alternative (https://github.com/espressif/esp-idf/blob/master/components/spiffs/spiffsgen.py). However I can still review and merge PRs, should you find some bug!

sfranzyshen, 2020-01-30:
I tried what you suggested and from the command line it worked fine ... with both of the versions (the one from esp8266 arduino distribution and from your releases) I reinstalled arduino 1.8.10 and esp8266 core 2.6.3 and tried again ... this time everything worked great! thank you and sorry for taking your time :)

closed 2020-01-30 by sfranzyshen