igrr/mkspiffs #73: Windows build error

kapteynpeter, 2020-03-03:
I get the following error when trying to build:

```
$ mingw32-make dist
find: paths must precede expression: `./spiffs/LICENSE'
find: possible unquoted pattern after predicate `-path'?
g++ -std=gnu++11 -Os -Wall   -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-6-g983970e\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__   -c -o main.o main.cpp
cc -std=gnu99 -Os -Wall -mno-ms-bitfields  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-6-g983970e\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__   -c -o spiffs/src/spiffs_cache.o spiffs/src/spiffs_cache.c
<builtin>: recipe for target 'spiffs/src/spiffs_cache.o' failed
process_begin: CreateProcess(NULL, cc -std=gnu99 -Os -Wall -mno-ms-bitfields -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-6-g983970e\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -c -o spiffs/src/spiffs_cache.o spiffs/src/spiffs_cache.c, ...) failed.
make (e=2): The system cannot find the file specified.
mingw32-make: *** [spiffs/src/spiffs_cache.o] Error 2
```

What is my next move here?

kapteynpeter, 2020-05-01:
I built with `make CC=gcc CXX=g++`

closed 2020-05-01 by kapteynpeter

WicherW, 2024-08-03:
I got: 
``` 
$make dist
g++ -std=gnu++11 -Os -Wall   -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-7-gf248296\" -D SPI                   FFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE                   __   -c -o main.o main.cpp
cc -std=gnu99 -Os -Wall -mno-ms-bitfields  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-7-gf2                   48296\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\"                    -D __NO_INLINE__   -c -o spiffs/src/spiffs_cache.o spiffs/src/spiffs_cache.c
cc -std=gnu99 -Os -Wall -mno-ms-bitfields  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-7-gf2                   48296\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\"                    -D __NO_INLINE__   -c -o spiffs/src/spiffs_check.o spiffs/src/spiffs_check.c
cc -std=gnu99 -Os -Wall -mno-ms-bitfields  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-7-gf2                   48296\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\"                    -D __NO_INLINE__   -c -o spiffs/src/spiffs_gc.o spiffs/src/spiffs_gc.c
cc -std=gnu99 -Os -Wall -mno-ms-bitfields  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-7-gf2                   48296\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\"                    -D __NO_INLINE__   -c -o spiffs/src/spiffs_hydrogen.o spiffs/src/spiffs_hydrogen.c
cc -std=gnu99 -Os -Wall -mno-ms-bitfields  -Itclap -Iinclude -Ispiffs/src -I. -D VERSION=\"0.2.3-7-gf2                   48296\" -D SPIFFS_VERSION=\"0.3.7-5-gf5e26c4\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\"                    -D __NO_INLINE__   -c -o spiffs/src/spiffs_nucleus.o spiffs/src/spiffs_nucleus.c
g++ main.o spiffs/src/spiffs_cache.o spiffs/src/spiffs_check.o spiffs/src/spiffs_gc.o spiffs/src/spiff                   s_hydrogen.o spiffs/src/spiffs_nucleus.o -o mkspiffs.exe -Wl,-static -static-libgcc -static-libstdc++
strip mkspiffs.exe
cp mkspiffs.exe mkspiffs-0.2.3-7-gf248296-generic-win32/
zip -r mkspiffs-0.2.3-7-gf248296-generic-win32.zip mkspiffs-0.2.3-7-gf248296-generic-win32
make: zip: Polecenie nie znalezione
make: *** [Makefile:103: mkspiffs-0.2.3-7-gf248296-generic-win32.zip] Błąd 127
```
I tried with this:
https://github.com/nsfilho/E12AIO3/tree/master/tools
but i got the same error.

finally helped me your hint:
```
 make CC=gcc CXX=g++
 ``` 

Thanks a lot :)
by the way I was guided by this topic: https://github.com/igrr/mkspiffs/pull/81 and https://github.com/espressif/ESP8266_RTOS_SDK/issues/896