igrr/mkspiffs #79: Unable to unpack image created by spiffsgen

AlexanderKokushkin, 2021-02-07:
Hello, 

I'm creating spiffs image this way:
 spiffsgen.py 40960 --page-size 256 --block-size 4096 mypackdir image.bin
then unpacking with mkspiffs:
 mkspiffs.exe -b 4096 -p 256 -s 40960 -u mydestdir image.bin
and getting random numbers in extracted files.

The same issue is with the images downloaded from the chip. 
ESP-IDF version is 4.1. 
mkspiffs.exe correctly unpacks images created by himself. 
e.g. mkspiffs.exe -b 4096 -p 256 -s 40960 -c mypackdir image.bin
The resulting images created by mkspiffs and spiffsgen are different.
I tried several versions from /Releases with the same results.

Please help, maybe I missed some crucial parameter? Thanks.

Skizy, 2021-02-16:
Hi
I had the same behaviour. But on my ubuntu linux mkspiffs just crushed without attempt to create any of files.
Recently I've upgraded ESP-IDF to 4.1.1 (as well as toolchain to esp-2020r3) and now mkspiffs unpacks files perfectly!
Tried to generated image and image got from esp-wroom32, all are fine.

argilo, 2021-05-02:
spiffsgen.py uses a file metadata length of 4 by default, so I think you would need to build mkspiffs with `make dist CPPFLAGS="-DSPIFFS_OBJ_META_LEN=4"` to read such files.

Calebe94, 2021-05-22:
> spiffsgen.py uses a file metadata length of 4 by default, so I think you would need to build mkspiffs with `make dist CPPFLAGS="-DSPIFFS_OBJ_META_LEN=4"` to read such files.

Thank you @argilo. 
Building `mkspiffs` with `DSPIFFS_OBJ_META_LEN=4` worked for me.