igrr/mkspiffs #84: Make fails on M1 pro Mac book Pro

dejavu1987, 2023-03-31:
I am trying to build on M1 mac book pro. I get the following error.
```
      _spiffs_gc_find_candidate in spiffs_gc.o
      _spiffs_object_append in spiffs_nucleus.o
      _spiffs_object_truncate in spiffs_nucleus.o
  "_mkdir", referenced from:
      dirCreate(char const*) in main.o
  "_opendir$INODE64$UNIX2003", referenced from:
      addFiles(char const*, char const*) in main.o
      dirExists(char const*) in main.o
  "_printf", referenced from:
      _SPIFFS_vis in spiffs_hydrogen.o
  "_putchar", referenced from:
      _SPIFFS_vis in spiffs_hydrogen.o
  "_puts", referenced from:
      _SPIFFS_vis in spiffs_hydrogen.o
  "_readdir$INODE64", referenced from:
      addFiles(char const*, char const*) in main.o
  "_stat$INODE64", referenced from:
      addFiles(char const*, char const*) in main.o
  "_strcmp", referenced from:
      addFiles(char const*, char const*) in main.o
      _spiffs_object_find_object_index_header_by_name_v in spiffs_nucleus.o
      _spiffs_obj_lu_find_free_obj_id_bitmap_v in spiffs_nucleus.o
      _spiffs_obj_lu_find_free_obj_id_compact_v in spiffs_nucleus.o
  "_strlen", referenced from:
      addFile(char*, char const*) in main.o
      addFiles(char const*, char const*) in main.o
      listFiles() in main.o
      unpackFiles(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) in main.o
      std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>::basic_string[abi:v15006]<std::nullptr_t>(char const*) in main.o
      _SPIFFS_creat in spiffs_hydrogen.o
      _SPIFFS_open in spiffs_hydrogen.o
      ...
  "_strncpy", referenced from:
      _spiffs_stat_pix in spiffs_hydrogen.o
      _spiffs_object_create in spiffs_nucleus.o
      _spiffs_object_update_index_hdr in spiffs_nucleus.o
  "start", referenced from:
     implicit entry/start for main executable
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [mkspiffs] Error 1
```

pawelskr, 2023-08-09:
A Makefile seems to be outdated. 
I used a `dirty hack` and commented out this section in Makefile and then I managed to build tool without any errors 😄 
```
ifeq ($(TARGET_OS),osx)
	TARGET_CFLAGS   = -mmacosx-version-min=10.7 -arch i386 -arch x86_64
	TARGET_CXXFLAGS = -mmacosx-version-min=10.7 -arch i386 -arch x86_64 -stdlib=libc++
	TARGET_LDFLAGS  = -mmacosx-version-min=10.7 -arch i386 -arch x86_64 -stdlib=libc++
endif
```

new-soul, 2023-08-11:
`ifeq ($(TARGET_OS),osx)
	TARGET_CFLAGS   = -mmacosx-version-min=10.7 -arch x86_64
	TARGET_CXXFLAGS = -mmacosx-version-min=10.7 -arch x86_64 -stdlib=libc++
	TARGET_LDFLAGS  = -mmacosx-version-min=10.7 -arch x86_64 -stdlib=libc++
endif`
I remove the '-arch i386' is OK now.

descuidado, 2025-01-27:
Please add arm64 support

razvanphp, 2025-05-26:
You can also compile it with `-arch arm64`, just change the Makefile.